kerberos-io/kios #1: KIOS buildroot configuration

cedricve, 2016-02-24:
@ccrisan this is a follow-up ticket of kerberos-io/machinery#30, as suggested these are the tasks to be completed.

- [Stripped version of motionEyeOS]
- [Install package for web]
- [Install package for machinery]
- [Testing]
- [Publish to master]

ccrisan, 2016-02-28:
Alright. I pushed the promised code to the `develop` branch. It can be built by following [motionEyeOS's instructions](https://github.com/ccrisan/motioneyeos/wiki/Building-From-Source). The resulting image is a 220MB uncompressed file (~36MB gzipped) that can be written directly to the SD card.

The first boot will most likely take 1-2 minutes on a PI1, but subsequent boots will be significantly faster. A connected monitor will show a clean boot log.

If an ethernet connection is detected, DHCP is used to configure it. WiFi adapters are ignored unless explicitly configured (I really need to create a wiki page on this matter!).

Services that are included with this version and that will automatically run at boot:

 * busybox-based syslog
 * watchdog feeder
 * brownout detector
 * (e)udev
 * date updater (via an HTTP HEAD request)
 * crond
 * sshd (root, no password)
 * proftpd
 * samba

cedricve, 2016-02-28:
Crisan, nice! This is a huge step forward for our community. Making video surveillance available for everyone.. I will try to build the project on a VM, after this we can discuss creating the packages for both machinery and the web repositories. I think the machinery will be very easy, as it only has FFMPEG as a dependency; just need to cmake and make the project. The web will be more difficult, as it requires a webserver (prefer nginx) and PHP, I will need your help on this.

Some questions:

- Why does the system boots 1-2 minutes the first timer? What is it doing?
- WIFI connection is a MUST-HAVE, how difficult is it to configure/install? Don't know if we have time to build an installer which is CROSS-platform? 

Thanks for all your hard work, this is highly appreciated!

ccrisan, 2016-02-28:
The first boot will prepare the system by creating the data partition (with its required directory structure). The most time-consuming task at this step creating the SSH keys. If you'll plug in a monitor, you'll see the SSH service taking forever at first boot. With a RPI2 (or any other board for that matter), things are a lot faster.

I know WiFi is a must have and I never thought otherwise :) But as you probably know, motionEyeOS relies on an extended motionEye UI for all system configuration, including WiFi and static IP configuration. Until we figure out how to do this from UI, you could simply use the `writeimage.sh` script to preconfigure WiFi when writing the image. See [this](https://github.com/ccrisan/motioneyeos/wiki/Installation) for more details.

In the mean time I'll try to take some time and document all the config files used by motiionEyeOS to configure various system parts. From that point on, we can decide how exactly these files will be (pre)configured.

cedricve, 2016-02-28:
Thanks for the explanation, that makes sense. The problem I see with the bash script is that a lot of users are using Windows. It's ok for the flashing part with Win32Imager, but the writeimage.sh will of course not work for them.

For this reason I've [used a different approach](https://kerberosio.zendesk.com/hc/en-us/articles/202817502-How-to-connect-to-WIFI-). Isn't this something we can use for our systems? Everyone can open and edit a file.

ccrisan, 2016-02-28:
Of course that's doable. But the best approach in my opinion is to prepare a customized image right from a web page that allows you to download it.

cedricve, 2016-02-29:
Hmm sounds good, we need to find a way to generate the image server-side and deliver it to the end users. It would be good if we would have a GUI (on a webpage) which let's a user select an existing WIFI connection; don't know if there are any libraries for this?

I've just created (879ae77baddb98fb160f39ae20ecb9fa45034111) two packages "web" and "machinery", I will experiment further with it..

ccrisan, 2016-02-29:
I could build a web service that handles all that. What I won't do is build a UI for it - I find it terribly difficult and tiresome. If there's anyone willing to wrap a UI around a restful/json/ajax API for that, then we have the solution.

cedricve, 2016-02-29:
Sure if you can create an API, then I'll help with the UI part.  Are they any other parameters required  then network properties?

ccrisan, 2016-02-29:
There are some advanced settings as well. I'll make a list. Or a mockup or something.

cedricve, 2016-03-11:
Machinery is succesfully compiled on the develop branch. Several patches where created to make it possible to cross-compile. A startup file has been added, which will start the machinery on boot.

cedricve, 2016-03-13:
Web is succesfully compiled. PHP (+extension) and NGINX is added, the web repository is compiled on the host and copied to the target library. Next is to fix the wireless connection.

cedricve, 2016-04-01:
Testing finishing, now packaging and updating documentation. Expected release is 1 May 2016.

closed 2016-04-21 by cedricve