kerberos-io/kios #6: security issue: memcached is listening on all interfaces by default

EmPeWe, 2016-10-18:
According to https://github.com/memcached/memcached/wiki/ConfiguringServer the recommendation is: "...you must not expose memcached directly to the internet..."

It would be a better approach to start the daemon with -l 127.0.0.1 or even better using unix sockets (-s <file>), which is supported in php-memcached since version 2.0.0b1

cedricve, 2016-10-18:
thanks for the comment, we will integrate this in the next release.

natefanaro, 2019-09-26:
Is this still an issue? I have not tried this yet but according to the init script of memcached this wasn't resolved https://github.com/kerberos-io/kios/blob/master/board/common/overlay/etc/init.d/S63memcached#L4

Will this be fixed? I am trying to evaluate if I want to use this software. I generally trust the devices on my network and know not to expose this to the public internet. Still, seeing an open security issue this old is a concern.