kerberos-io/kios #7: images for easy installation / testing on x86 / amd64 machines

mark-orion, 2017-03-21:
I think this project would benefit from kios images for x86 / amd64 systems. This would allow very easy testing or even using of kerberos-io on "standard" desktops and laptops by using virtualbox or a USB image. I am aware of people who do have an "old" computer sitting around that could do the job, but who do not want to buy a Raspberry Pi.

cedricve, 2017-03-21:
hey @mark-orion correct, this is the reason why we also ship Docker images. https://doc.kerberos.io/2.0/installation/Docker

mark-orion, 2017-03-21:
Yes I saw this, but for testing you probably want to use a USB camera and not an IP based one.

cedricve, 2017-03-22:
Well @mark-orion I hope this will be possible very soon. Don't know how far it progressed, so you can access an USB camera from a host system like OSX or Windows. Linux shouldn't be any problem.

http://stackoverflow.com/questions/41023827/accessing-usb-webcam-hosted-on-os-x-from-a-docker-container

https://forums.docker.com/t/how-to-share-an-non-storage-usb-device-between-host-and-container-on-osx/9385

http://apple.stackexchange.com/questions/265281/using-webcam-connected-to-macbook-inside-a-docker-container

A possible answer is stream the camera into the container using ffmpeg according to a comment in http://stackoverflow.com/questions/37960828/webcam-streaming-from-mac-using-ffmpeg

closed 2018-06-06 by cedricve