kerberos-io/kios #8: Raspberry Pi Zero W support

Pilskalns, 2017-03-22:
Hi,

Tried Raspberry Zero KiOS image on Zero W but it even doesn't boot or blink any image. When I plug same SD card into regular Zero, it boots and works with USB wi-fi dongle _out of the box_. Also, my Zero W works with other distros.

So, I suppose fault is how SD card is prepared. What are possibilities to get it working on Zero W? :)

cedricve, 2017-03-22:
Hey @Pilskalns, we're preparing a seperate release to support Raspberry Pi Zero W. I will be available in a few days..

matright, 2017-04-04:
Hi,
I'm also interested in this, do you have any news ?

Thank you

cedricve, 2017-04-24:
Hey all,

Can you please try this release: https://github.com/cedricve/kios/releases/download/develop/kios-raspberrypi-20170423.img.gz

You can follow the advanced installation https://doc.kerberos.io/2.0/installation/KiOS#advanced

Cédric

cmaaser, 2017-04-24:
Doesn't work for me. I only get the RGBC test image shown when powering up my Pi Zero W. The boot partition does not contain a 'bcm2708-rpi-0-w.dtb' file like e.g. raspbian jessie, which might be the cause.

cedricve, 2017-04-25:
hey @cmaaser sorry I did a wrong build..  Can you retry with this one.

https://github.com/cedricve/kios/releases/download/develop/kios-raspberrypi-20170425.img.gz

cmaaser, 2017-04-25:
This build doesn't seem to boot either @cedricve . It is stuck again on the colored splash screen.

cedricve, 2017-04-25:
@cmaaser do you see the bcm2708-rpi-0-w.dtb file in /boot?

cmaaser, 2017-04-25:
@cedricve Yes I do.

cmaaser, 2017-04-25:
When replacing start.elf, fixup.dat and bootcode.bin from the latest raspbian image, which have significantly different file sizes, my pi0w boots up (but prints "Failed to create camera component ...." later on).

cedricve, 2017-04-25:
Great.. well that's because you didn't connect a raspberry pi camera. Can you access the web interface? I'm just reviewing the build process and it looks like I didn't upgraded the userland and firmware libraries. Cross-compiling right now. Thanks for helping!

cmaaser, 2017-04-25:
Actually I do have a camera connected, but it is a new one which I haven't tested before on one of my other regular PIs due to lack of time. I'll test the camera and web interface tomorrow.

cedricve, 2017-04-25:
Could also be because of the userland library. I'll drop a new build in this topic in a few minutes.

cedricve, 2017-04-26:
@cmaaser updated the image again, let me know if you had time to check it out. 
https://github.com/cedricve/kios/releases/download/develop/kios-raspberrypi-20170425.img.gz

vincenzocostagliola, 2017-04-26:
Hi @cedricve , 
the latest image seems working. 
I'm working with the RPI camera and kerberos cloud at moment.
Let me know if you need more info.

cedricve, 2017-04-26:
@vincenzocostagliola did you tried the one I've commented above? Can you confirm the RPI camera is also working?

vincenzocostagliola, 2017-04-26:
@cedricve Yes I've tried the image that you posted 3 hours ago.
Yes the RPI Camera ( Rev 1.3 Model) is working.

vincenzocostagliola, 2017-04-26:
@cedricve I've just tried the v2 Pi NoIR on another PI Zero W and it's working flawlessly.

cedricve, 2017-04-26:
Awesomeness @vincenzocostagliola ! Ok let's release this. I'll update the docs and created a new release. Can you also verify everything works as expected (e.g. the web interface and all it's pages). I just ordered a Pi Zero W yesterday.

vincenzocostagliola, 2017-04-26:
@cedricve Seems everything is fine. I've configured the motion and the cloud with no problem. System page also works as expected.

cmaaser, 2017-04-26:
I can confirm that the latest image works flawlessly including camera (V2.1), web access and wifi. Thanks @cedricve !

cedricve, 2017-04-26:
Awesome guys, let's release the beast. Thank you all for testing and helping.

cedricve, 2017-04-26:
Released https://github.com/kerberos-io/kios/releases/tag/v2.3.0

closed 2017-04-26 by cedricve

paulbakker, 2017-07-31:
There are 3 images in the releases (kios-raspberrypi-20170425.img.gz, 98.6 MBkios-raspberrypi2-20170425.img.gz, 98.6 MBkios-raspberrypi3-20170425.img.gz). Which one is the correct one for the pi zero?

cedricve, 2017-07-31:
@paulbakker kios-raspberrypi-20170425.img.gz is the correct one.