kerberos-io/kios #12: mac installer dropdown does not list version

kydouglas, 2017-06-11:
(kios darwin on mac 10.12.5) 
etcher fork does not display version on dropdown menu when trying to make images for sd card

https://ibb.co/geyskv

How to I load .img or is this a bug?

cedricve, 2017-06-11:
Might take a couple of seconds before it's loaded..

cedricve, 2017-07-16:
we'll update the current installer.

badams, 2017-08-13:
@cedricve any work arounds for this in the mean time? pretty keen to try kerberos out, it looks great. but the advanced installation guide is missing from the docs and currently the mac installer doesn't seem to work due to this issue.

cedricve, 2017-08-13:
hey @badams how do you mean is missing?

badams, 2017-08-13:
@cedricve  The advanced installation button is greyed out and doesnt link anywhere

[image]

btw, thanks for the quick response :)

cedricve, 2017-08-13:
No problem, just curious which browser you're using? Works on chrome.

badams, 2017-08-13:
Ah I see, looks like a js issue. I'm also using Chrome, I found i can just switch the `display:none` off in the web inspector to view the guide.

I'm getting some errors in the console 

[image]

Looks to be related to this 500 error

[image]

cedricve, 2017-08-13:
That's weird can you clear cache or so? What happens if you open this file directly http://doc.kerberos.io/js/vendor/requirejs/require.js

badams, 2017-08-13:
Cleared cache, and tried to access that file directly;  and got `ERR_TOO_MANY_REDIRECTS`

cedricve, 2017-08-13:
Can you try once more @badams disabled our CDN.

badams, 2017-08-13:
Still the same @cedricve, not sure if this is helpful but.. accessing `https://doc.kerberos.io/js/vendor/` shows a stacktrace stating

```
The "/var/www/kerberos-public/docs/public/documentation/js/" directory does not exist.
```

cedricve, 2017-08-13:
@badams ok should be fixed now @badams. Indeed dependenices wre missing.

badams, 2017-08-13:
Working now, thanks! also sorry to hijack this issue @kydouglas

cedricve, 2017-08-13:
no problem, great you reported it! I didn't was aware as i stilled the existing files in the CDN (and not a fresh copy). Regarding the installer did you already tried to restart it? I've seen the issue with not loading the versions before, but I resolved it by restarting it.

badams, 2017-08-13:
@cedricve no problem at all, at this rate (assuming kerberos works well) I'll be signing up for the cloud service to manage a few instances.

Yes I tried restarting and reinstalling the installer, also tried leaving it open for a while incase it was downloading in background.  would be useful if I could turn on debug mode or find some error logs.

TheButterZone, 2018-06-24:
How do I get the Mac KiOS installer (_Select a version_ dropdown empty) to see https://github.com/kerberos-io/kios/releases/download/v2.7.1/kios-raspberrypi3-20180607.img.gz (downloaded) so I can do the GUI configuration before flashing with it?