kerberos-io/kios #19: Password set at every reboot

iooner, 2017-07-31:
Hi!
At every reboot, Kios ask to set the password.

cedricve, 2017-07-31:
Not sure what you mean, can you supply a screenshot?

badams, 2017-08-13:
Ran into this too.

I unplugged my RPi  and plugged back in, but when I tried to log into Kerberos i was asked to create a user account again.

This seems to be caused by [this line](https://github.com/kerberos-io/web/blob/283f053a8e975f18279b2608478347ee4ad029d9/routes/web.php#L26)

My understanding is that when the app is restarted the config values are just read from [this file](https://github.com/kerberos-io/web/blob/master/config/kerberos.php#L4) which leads to the account setup page.

cedricve, 2017-08-13:
thanks for reporting @badams, well the config should be updated when finishing the welcome page.
https://github.com/kerberos-io/web/blob/283f053a8e975f18279b2608478347ee4ad029d9/app/Http/Controllers/UserController.php#L38-L58

Not sure what is going wrong there.

cedricve, 2017-08-13:
@badams just tried it myself using v2.4.1 but couldn't reproduce. Can you try it again with a fresh install?

cedricve, 2017-11-01:
Will be fixed in next release.

cedricve, 2017-11-16:
fixed in KiOS v2.5.0

closed 2017-11-16 by cedricve