kerberos-io/kios #24: Kios 2.5.0 update has broken kios

traitors-gate, 2018-01-10:
(5/5) ErrorException
Undefined index: file (View: /var/www/web/resources/views/settings_controls.blade.php) (View: /var/www/web/resources/views/settings_controls.blade.php) (View: /var/www/web/resources/views/settings_controls.blade.php) (View: /var/www/web/resources/views/

traitors-gate, 2018-01-20:
Had time to try rebuilding my camera setup from scratch. Everything worked great until I uplifted to 2.5.0. I now have same issue as originally reported. The KIOS configuration page on the GUI throws a 5/5 ErrorException on a Mac using safari. Install is on a pi 3 and the camera is IP, network is wired ethernet.

cedricve, 2018-01-20:
hey @traitors-gate, can you send me the system files? I think your capture.xml config file is corrupted, while upgrading.

traitors-gate, 2018-01-21:
I have fixed it by installing the new version directly onto the SD card through the installer.

For some reason the upgrade through the kios UI corrupts the config file. I tried 3 times and the same thing happened.

Sorry but I don’t have a copy of the corrupted files - I only had one SD card and I had to fully reformat it to get the kios installer to work.

Thanks

Adrian

cedricve, 2018-01-22:
we'll fix this in next release.

cedricve, 2018-01-22:
@traitors-gate can you send me your current logs (although they work properly now).

traitors-gate, 2018-01-24:
Sorry, I will not have time to pull the logs until the weekend. I noticed that you referenced a camera url bug (#17) - here is my camera URL if that helps:

http://admin:password@10.10.10.200/snapshot.jpg?user=admin&pwd=password&strm=0

cedricve, 2018-02-02:
will be fixed in version 2.6.0

closed 2018-02-02 by cedricve