kerberos-io/kios #27: KiOS isn't storing configuration settings and continues reverting to the wizard

espressobeanies, 2018-02-15:
OS: KiOS v2.6.0
Platform: Raspberry Pi 2

Issue: KiOS isn't storing login credentials generated during the initial wizard setup and is re-prompting the wizard after a set number of minutes. Once the wizard completes again, the initial settings reappear.

cedricve, 2018-02-15:
Weird should be fixed by now. Which browser are you using? Is this a fresh install?

espressobeanies, 2018-02-20:
Mozilla Firefox and yes it was. I only saw the issue occur once so far, but
I haven't tried logging into it again recently.

espressobeanies, 2018-03-07:
Resolved in v2.6.0 via /data/etc/kios.conf file

closed 2018-03-07 by espressobeanies

cedricve, 2018-03-07:
hey @espressobeanies, how do you mean resolved with /data/etc/kios.conf?
Can you explain your research?