kerberos-io/kios #31: wpa_supplicant not starting on RaspberryPi 3 B (no device)

marcel31415, 2018-05-26:
Hi!

Updating via webinterface from 2.6.1 to 2.7.0 failed, so I reflashed it (Etcher).
But the Pi did not appear in my local network. Checking the boot process on my monitor exposed the problem: wpa_supplicant does not start as it does not find a device.
Flashed it two times, tried two different Rasp 3 Bs. Flashing 2.6.1 works.

cedricve, 2018-05-26:
thanks @marcel31415, I disabled 2.7.0 temporary as I'll resolve the issue first before other people are affected. keep you posted.

cedricve, 2018-05-27:
It looks like something is changed since latest raspbian kernel release. The problem is also appearing on the Raspberry Pi Zero W. So there might changed something for the onboard wifi.

cedricve, 2018-05-28:
Hey @marcel31415, we've updated the image. Can you try again?

marcel31415, 2018-05-28:
I've just updated via web-interface and the RaspPi 3 came back online. Seems to work properly. Thank you!

closed 2018-05-28 by cedricve