kerberos-io/kios #33: livestrem fails on dashboard when all requests are made through https

ricardolsfranca, 2018-07-20:
I've been testing kerberosio in a remote installation connected to the internet through a 4G modem. All the traffic passes through a VPN server that redirects all requests to the machine running kerberosio. Recently, I've configured **letsencrypt** on **nginx** for obtaining a Qualys A security configuration. When setting all requests to be redirected to https, the live stream on the dashboard doesn't load anymore. 

I'm guessing that the code tries to access the port 80 and the process fails with the redirecting.

cedricve, 2018-07-21:
Hmm interesting, can you provide some details about the nginx config @ricardolsfranca ?

ricardolsfranca, 2018-07-23:
Here's the configuration file. Maybe it's a misconfiguration.

`server
{
    listen 80 default_server;
    listen [::]:80 default_server;
    root /var/www/web/public;
    server_name kerberos.rpi;
    index index.php index.html index.htm;
    location /
    {
            autoindex on;
            try_files $uri $uri/ /index.php?$query_string;
    }
    location ~ \.php$
    {
            fastcgi_pass unix:/var/run/php/php7.0-fpm.sock;
            fastcgi_param SCRIPT_FILENAME $document_root$fastcgi_script_name;
            include fastcgi_params;
    }
}

server
{
    root /var/www/web/public;
    server_name logan.zefiroengenharia.com; # managed by Certbot
    index index.php index.html index.htm;
    location /
    {
            autoindex on;
            try_files $uri $uri/ /index.php?$query_string;
    }
    location ~ \.php$
    {
            fastcgi_pass unix:/var/run/php/php7.0-fpm.sock;
            fastcgi_param SCRIPT_FILENAME $document_root$fastcgi_script_name;
            include fastcgi_params;
    }

    listen [::]:443 ssl ipv6only=on; # managed by Certbot
    listen 443 ssl; # managed by Certbot
    ssl_certificate /etc/letsencrypt/live/logan.zefiroengenharia.com/fullchain.pem; # managed by Certbot
    ssl_certificate_key /etc/letsencrypt/live/logan.zefiroengenharia.com/privkey.pem; # managed by Certbot
    include /etc/letsencrypt/options-ssl-nginx.conf; # managed by Certbot
    ssl_dhparam /etc/letsencrypt/ssl-dhparams.pem; # managed by Certbot

}

server
{
    if ($host = logan.zefiroengenharia.com) {
        return 301 https://$host$request_uri;
    } # managed by Certbot

    listen 80 ;
    listen [::]:80 ;
    server_name logan.zefiroengenharia.com;
    return 404; # managed by Certbot

}
`