kerberos-io/kios #37: kios "configuration" ui never loads on Pi Zero W

stenwt, 2018-10-20:
On both an installation of Kios v2.7.2 that's been running for a few days, and a new installation with no configuration applied, I'm getting the "configuration" screen hanging/never loading [image]
. System and dashboard load fine. The system loadavg is roughly 1, logs don't show any issues, and the system is otherwise responsive. 
[kerberosio-configuration.tar.gz](https://github.com/kerberos-io/kios/files/2498492/kerberosio-configuration.tar.gz)

xhozt, 2020-02-19:
confirmed here .

cedricve, 2020-02-19:
@xhozt which version?

grenade, 2020-06-29:
happens for me too pretty constantly on pi 3 and 4 as well (kios version 2.8.0).
issue happens after ip address changes caused by power outages. only workaround i have found is to reflash the sd card. had to buy a better rut955 router to lock the dhcp leases to the cameras and kios instances to save me having to go and reflash kios every time we have a power cut to the router, camera or kios device.

cedricve, 2020-06-29:
hmm that is weird @grenade , what do you see in the web logs?

cedricve, 2020-06-30:
@grenade  let me know if you want to elaborate on this issue.

grenade, 2020-07-01:
I will upload logs when I get the chance. I have a couple sdcards from borked kios instances.

cedricve, 2020-07-01:
Thank you I would be great to get this sorted out.