kerberos-io/kios #38: kerberos-io keeping trying to start machinery

nopub, 2019-02-11:


nopub, 2019-02-11:
Hi,

I'm trying to setup a PI3B surveillance system and used KiOS-linux-x64.AppImage to flash an SD card.
When booting the Raspberry the system keeps looping and displaying the same messages again and again (difficult to capture). I have to admit that I'm pretty new to Raspberry and Linux.

After booting the Raspberry with the flashed SD card, I always see the following lines.

.
.
OMX_GetHandle(OMX.broadcom.camera) failed 80001000
[OMX.broadcom.video.encode] SetParameter OMX_IndexParameterPortDefinition failed 80001005
line 24 1559 segmentation fault kerberos.io --config $conf
Starting kerberos.io machinery
.
.

Nothing else is happening.

Did I do something wrong?

Thanks for your help,

Fabrizio

cedricve, 2019-02-12:
Hello make sure you have selected the right camera. By default is uses the Raspberry Pi camera.

Cédric

nopub, 2019-02-12:
Hi,
I thought I could change it onces kios had booted. I'll change the config and capture xml (for IPCamera and rtsp address) on the SD card and retry.
Fabrizio

bennybubble, 2019-03-15:
Same issue here, I've loaded the latest image on a card and rebooted. On the monitor it keeps scrolling and complaining like above...
However I do not have a camera connected yet (plan to use a USB camera), how do I get around this?
Which config do I need to edit (after etching) so it will stop doing this ?

nopub, 2019-03-17:
I stopped trying kerberosios and did it with raspbian. Works fine but one question -> I'll open another post.
Thanks for the nice product.

closed 2019-03-17 by nopub

cedricve, 2019-03-17:
@bennybubble by default the camera expects the raspberry pi camera, if not the system will try over and over until it recognises it.. that is what you see when you start up your raspberry pi when attached to an external screen.

What you need to do is change the config through the web interface or SSH into the Raspberry Pi, and change the config.xml file.