kerberos-io/kios #42: Installer does not work on Arch Linux

avanc, 2019-10-01:
I get the following error while starting the Appimage:

```
/tmp/.mount_PgaY91/usr/bin/kios.wrapper
kdialog: Unknown options: caption, y, e, s, n, o.

(kios:6693): Pango-ERROR **: 19:35:42.390: Harfbuzz version too old (1.2.1)

Trace/breakpoint trap (core dumped)
```
I'm using Arch Linux which I updated today.

avanc, 2019-10-01:
Unfortunately, the documentation for advanced installation is missing:
https://doc.kerberos.io/2.0/installation/installation/KiOS#advanced

rogaven2002, 2019-11-28:
I found kios today and downloaded it.
I'm using PCLinuxOS 64bit. Same error message.
installed pango 1.44.7 and harfbuzz 2.6.4
What can I do?
I would like to test the program.