kerberos-io/kios #44: Installer does not work on version 2.8.0 with WiFi setup RPi3+

kbabka, 2020-01-05:
Etcher installer frozen by downloading 100% and nothing happen.

When chose wired configuration, its O.K.

cedricve, 2020-01-05:
Thanks for reporting, is it only when selecting WIFI? Or also with the Static IP?
Which OS are you using?

Cédric

kbabka, 2020-01-05:
Windows 10, dynamic IP

cedricve, 2020-01-05:
Interesting just received another ticket about the same issue. Would you have time to quickly verify if you use the combo (ethernet + static IP)?

Can confirm the downloading works, but once you reach 100%, it stops working.

kbabka, 2020-01-05:
I tested ethernet + static IP and also installer frozen on 100%.

cedricve, 2020-01-05:
Thanks the issue is here: https://github.com/kerberos-io/etcher/blob/master/lib/gui/app/modules/image-writer.js#L389.

Will need to have some time to debug, but we will get there. Thank for reporting.

cedricve, 2020-01-07:
@kbabka we fixed the issue, can you please retry by downloading the installer again, and execute it.

kbabka, 2020-01-07:
Now installer freezes by "Starting ..." see picture

[image]

cedricve, 2020-01-08:
How long did you wait might take some time before the changed are applied during the starting process.

Kind regards,
Verstraeten Cédric

kbabka, 2020-01-08:
I waited about half an hour, maybe longer.
Regards
Karel

kbabka, 2020-01-08:
When I use original Etcher and download Kerberos.io image file version 2.8 for RPi3, put it in SD card and then copy the configuration files to the first partition, everything works including WiFi connection

cedricve, 2020-01-08:
You are right, the build is not correct. Will come back.

cedricve, 2020-01-09:
@kbabka, fixed and tested myself. Please give me your confirmation as well. Thanks!

kbabka, 2020-01-09:
Yes, I confirm that the installation is already OK. 
Thanx.

closed 2020-01-09 by kbabka