kerberos-io/kios #45: Cannot initializate device with 2.8.0

peskymerryherald, 2020-02-12:
[image]

After installing a kios 2.8.0 in a raspberry pi zero, I get stuck in select language page.

cedricve, 2020-02-12:
thanks for reporting, will try to reproduce. any help is welcome though.

cedricve, 2020-02-12:
@peskymerryherald  what happens if you open in incognito?

peskymerryherald, 2020-02-12:
@cedricve, same error 😞

peskymerryherald, 2020-02-18:
Same it's happening to me with a Raspberry Pi 4.

igor-roje, 2020-04-06:
Same it's happening to me with a Raspberry Pi 3 Model B v 1.2.

cedricve, 2020-04-06:
igor didnt experience this.. tried to reproduce. What happens if you do a clean install?

igor-roje, 2020-04-06:
I will tray clean install for now if this png can help
[image]

cedricve, 2020-04-06:
thanks looks different, let me know if you did the reinstall.

igor-roje, 2020-04-06:
Did you meen reinstall kios image or

Iinstall the Kerberos  manually on Raspbian GNU/Linux 10.

cedricve, 2020-04-06:
I assume you installed it via KIOS?

igor-roje, 2020-04-06:
Yes I am

igor-roje, 2020-04-06:
OK, now I understand :

WARNING: Upgrading from a previous to 2.8.0 will not work. Please do a clean install

I do a clean install the first time and I get that error.

igor-roje, 2020-04-06:
Maybe this Internal Server Error help :
[image]

cedricve, 2020-04-06:
can you show what is in the response of the call (check the network tab)

igor-roje, 2020-04-06:
I installed the Kerberos manually on Raspbian GNU/Linux
and everything works fine so I am sure that the hardware works properly.
Tomorrow I will back to Kios and send you the response.

igor-roje, 2020-04-07:
Here is the response.
[image]

cedricve, 2020-04-07:
hey @igor-roje, sorry cannot see your response.

igor-roje, 2020-04-07:
Sorry here it is.

igor-roje, 2020-04-07:
[response.txt](https://github.com/kerberos-io/kios/files/4445592/response.txt)

cedricve, 2020-04-08:
@igor-roje I just tested out myself what happens on the RPi 3 1.2, and it works just fine. So I'm wondering if we could look together during a video call? To help and understand what is happening?

igor-roje, 2020-04-08:
No problem what I need to do or install

cedricve, 2020-04-08:
Can you install teamviewer and send me your details (cedric@verstraeten.io <mailto:cedric@verstraeten.io>)

igor-roje, 2020-04-08:
Cedric,
Now it is working I dont no what you change

Igor

napisao je:

igor-roje, 2020-04-08:
Cedric,

I figured out the problem!

First, I did a clean install and the problem was the same.
Then I logged in Pi and canged the date ( date -s "2020-04-08 17:34").
After that I passed the installation with no problems!

It is interesting that when I reset  Pi he loses the date but it is not a
problem anymore because I passed the installation step.

The idea came to me when I saw "1970 year" when you openned some files on
my Pi (with 1970 year). That is the date by default I believe.

Just wanted to let you know!!

Best regards,
Igor

sri, 8. tra 2020. u 16:14 igor Roje <igor.roje@gmail.com> napisao je:

> Cedric,
> Now it is working I dont no what you change
>
> Igor
>
> napisao je:
>
>> Can you install teamviewer and send me your details (
>> cedric@verstraeten.io <mailto:cedric@verstraeten.io>)
>>
>> >
>> >
>> > No problem what I need to do or install
>> >
>> https://github.com/kerberos-io/kios/issues/45#issuecomment-610956706>,
>> >
>>
>>
>
>
> Igor Roje
>
> LITIJ d.o.o.
> Slavićeva 38
> 21000 Split
> HRVATSKA
> +385 98 359 111

cedricve, 2020-04-08:
Hmm that is strange, however great you find out! Not sure why the time sync is not working for you.
Are you within a highly secured network?

igor-roje, 2020-04-08:
No I don't think so, I am working from home these days.

napisao je:

igor-roje, 2020-04-08:
I did clean install one more time but this time I set static_dns="83.139.103.3" 
instead static_dns="192.168.0.1" in static_ip.conf which is my internet provider dns.
Now everything is working  fine.
Date and time are placed and installation is working fine.
Cedric thanks for your time.

cedricve, 2020-07-04:
Will check why it's not working properly without any internet accesss.

Fabeson, 2022-03-30:
I know this got a bit stale but since 2 years have passed without an update: @igor-roje fix to manually set the date via eg.
`date -s "2022-03-30 20:00"` is still valid and helped me out in getting through the initial setup - Thanks a lot!