kerberos-io/kios #46: Kios should not connect to an external MQTT server without permission

jeroenjacobs79, 2020-07-17:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

NA

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

NA
### Log Files

NA

### Actual issue

I'm seeing connections to an external MQTT server (159.65.215.225) after installing kerberos on my Raspberry Pi. Since this is data gathering without explicit permission from the user, this is probably a GDPR violation and therefore illegal in Europe.

litij, 2020-09-03:
Is there any workaround about this issue ?

cedricve, 2020-09-03:
hey, it's actually setting this connection up for our cloud application, butt if you didt connect it (configured it), it will not do anything. For next version we will make sure the connection is not established when not needed. However if you are suspicious you can always block the ip in the firewall.

jeroenjacobs79, 2020-09-03:
I had blocked those devices from connecting to the internet in my firewall, because there is absolutely zero reason it should connect to the internet without my permission.

@cedricve Sorry, but I don't think that's true. I never setup the cloud stuff, and it still connects to that MQTT server on the internet. I have my firewall logs to prove it. If someone ever logs a GDPR complaint about this, they probably have a case.