kerberos-io/kios #50: kerberosEtcher-1.5.70-x64.AppImage failing on ubuntu 20.04

tgelite, 2021-04-09:
This issue does not fall into the template in a logical way, so I'll put this preamble here:

I'm attempting to run kerberosEtcher-1.5.70-x64.AppImage on ubuntu 20.04

The gui comes up, lets me select version 2.8.0 to image with, and set up the network configuration, it also lets me select the SD card properly, but when I go to apply the image, it appears for a second to start, i see the promotional image and text on the left side of the window, and it indicates "starting", but then it asks me to log in and then switches to the screenshot below and is no longer imaging. Its unclear if its doing anything at first but after waiting for 1/2 an hour it never changes state.

[image]

How do I troubleshoot what is failing in the appimage?

I'm trying this on 
```
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=20.04
DISTRIB_CODENAME=focal
DISTRIB_DESCRIPTION="Ubuntu 20.04.2 LTS"
NAME="Ubuntu"
VERSION="20.04.2 LTS (Focal Fossa)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 20.04.2 LTS"
VERSION_ID="20.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=focal
UBUNTU_CODENAME=focal
```

### motionEyeOS Version

I am running KiOS version: (insert your version here, e.g. v2.8.0).
2.8.0

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).
raspberry pi zero, but my issue is with trying to run the installer on a laptop runnign ubunto 20.04

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My KiOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. kios.log)
 * (attach e.g. boot.log)

tgelite, 2021-04-09:
note when I sudo the launch of kerberosEtcher-1.5.70-x64.AppImage the same thing happens..

This is all I see in the console I launch it from command line:

```
agrajag@tor-XPS:~/Downloads$ sudo ./kerberosEtcher-1.5.70-x64.AppImage 
ready-to-show: 2663.468ms
```

and this is the syslogi output that comes up at the time of attempting to start imaging

```
Apr  8 22:31:05 tor-XPS kernel: [ 4681.609496] pcieport 0000:00:1c.4: AER: Corrected error received: 0000:00:1c.4
Apr  8 22:31:05 tor-XPS kernel: [ 4681.609501] pcieport 0000:00:1c.4: AER: PCIe Bus Error: severity=Corrected, type=Data Link Layer, (Transmitter ID)
Apr  8 22:31:05 tor-XPS kernel: [ 4681.609504] pcieport 0000:00:1c.4: AER:   device [8086:9d14] error status/mask=00001000/00002000
Apr  8 22:31:05 tor-XPS kernel: [ 4681.609505] pcieport 0000:00:1c.4: AER:    [12] Timeout               
Apr  8 22:31:20 tor-XPS kernel: [ 4696.832951] pcieport 0000:00:1c.4: AER: Corrected error received: 0000:00:1c.4
Apr  8 22:31:20 tor-XPS kernel: [ 4696.832959] pcieport 0000:00:1c.4: AER: PCIe Bus Error: severity=Corrected, type=Data Link Layer, (Transmitter ID)
Apr  8 22:31:20 tor-XPS kernel: [ 4696.832963] pcieport 0000:00:1c.4: AER:   device [8086:9d14] error status/mask=00001000/00002000
Apr  8 22:31:20 tor-XPS kernel: [ 4696.832966] pcieport 0000:00:1c.4: AER:    [12] Timeout               
Apr  8 22:31:24 tor-XPS kernel: [ 4700.662719] pcieport 0000:00:1c.4: AER: Corrected error received: 0000:00:1c.4
Apr  8 22:31:24 tor-XPS kernel: [ 4700.662726] pcieport 0000:00:1c.4: AER: PCIe Bus Error: severity=Corrected, type=Data Link Layer, (Transmitter ID)
Apr  8 22:31:24 tor-XPS kernel: [ 4700.662730] pcieport 0000:00:1c.4: AER:   device [8086:9d14] error status/mask=00001000/00002000
Apr  8 22:31:24 tor-XPS kernel: [ 4700.662732] pcieport 0000:00:1c.4: AER:    [12] Timeout               
Apr  8 22:31:26 tor-XPS kernel: [ 4702.647592] pcieport 0000:00:1c.4: AER: Corrected error received: 0000:00:1c.4
Apr  8 22:31:26 tor-XPS kernel: [ 4702.647600] pcieport 0000:00:1c.4: AER: PCIe Bus Error: severity=Corrected, type=Data Link Layer, (Transmitter ID)
Apr  8 22:31:26 tor-XPS kernel: [ 4702.647604] pcieport 0000:00:1c.4: AER:   device [8086:9d14] error status/mask=00001000/00002000
Apr  8 22:31:26 tor-XPS kernel: [ 4702.647608] pcieport 0000:00:1c.4: AER:    [12] Timeout               
Apr  8 22:31:34 tor-XPS gnome-shell[2599]: ../clutter/clutter/clutter-actor.c:10558: The clutter_actor_set_allocation() function can only be called from within the implementation of the ClutterActor::allocate() virtual function.
Apr  8 22:31:57 tor-XPS systemd[2194]: tmp-.mount_kerberi8hDsD.mount: Succeeded.
Apr  8 22:31:57 tor-XPS systemd[1]: tmp-.mount_kerberi8hDsD.mount: Succeeded.
```

lachlanbrewster1, 2021-04-28:
I have the exact same issue, left for about 30 minutes, no luck. The process itself seems to be _slowly_ writing something when I view it in system monitor (~5mb total).

Just flashed the plain image using standard balena etcher for now.

kr15h, 2022-02-10:
Same issue.