kerberos-io/kios #54: Kios fails to boot on a raspberry pi 4 B with 4 GB ram

maxvaneck, 2022-04-28:
When writing the image to the 12GB sd card everything works perfectly. but when booting the raspberry pi for the first time with the fresh install we get the notification. . unsupported software please download the latest version from https://www.raspberrypi.com. We tried finding our error since kios works perfectly fine with the old raspberry pi 3 we had on hand. We tried updating the eeprom but according to the utility it was allready the latest version. 
The kios version we tried was 2.8.0

kind regards,

 Max van Eck

damanbaird, 2022-05-08:
same here, maybe eeprom needs to be rolled back to about the date of the 2.8.0 release?

maxvaneck, 2022-05-11:
while that may fix the issue it does not address the underlying problem of kios being unable to handle newer bootloader versions

maxvaneck, 2022-05-12:
unortunately the rpi-eeprom-update tool rejects the 2019 eeprom due  to an illegal date error . 
so it seems that for the rpi 4 an update to kios will be necessary ehich  looking at te release cadence will be hopefully be somewhere in june/july.  i could install the flashing utility version from 2019 but at this point we're getting in such a mess of mismatched/old software versions that i'm starting to feel uncomfortable